IRNAS/PiRA-zero-firmware #9: Image resolution bug

SloMusti, 2017-12-07:
There is a bug in configuring image resolution, see:

```07.12.17 18:15:09 (+0100) Error while running processing in module 'pira.modules.camera'.
07.12.17 18:15:09 (+0100) Traceback (most recent call last):
07.12.17 18:15:09 (+0100)   File "pira/boot.py", line 194, in process
07.12.17 18:15:09 (+0100)     module.process(self.modules)
07.12.17 18:15:09 (+0100)   File "pira/modules/camera.py", line 61, in process
07.12.17 18:15:09 (+0100)     self._camera.resolution = self.resolution
07.12.17 18:15:09 (+0100)   File "/usr/local/lib/python2.7/site-packages/picamera/camera.py",line 2266, in _set_resolution
07.12.17 18:15:09 (+0100)     value = mo.to_resolution(value)
07.12.17 18:15:09 (+0100)   File "/usr/local/lib/python2.7/site-packages/picamera/mmalobj.py", line 400, in to_resolution
07.12.17 18:15:09 (+0100)     w, h = (int(i.strip()) for i in value.upper().split('X', 1))
07.12.17 18:15:09 (+0100)   File "/usr/local/lib/python2.7/site-packages/picamera/mmalobj.py", line 400, in <genexpr>
07.12.17 18:15:09 (+0100)     w, h = (int(i.strip()) for i in value.upper().split('X', 1))
07.12.17 18:15:09 (+0100) ValueError: invalid literal for int() with base 10: '2592\xd71944'
07.12.17 18:15:09 (+0100) ===============================================```

SloMusti, 2017-12-07:
Problem appears to be with may resolution above 1280x720, say  1920×1080, 2592×1944

kostko, 2017-12-07:
Did you copy these strings from somewhere? Your times (`x`) symbol seems like some strange unicode character, not an ordinary `x`. You should use an ordinary `x`.

SloMusti, 2017-12-07:
Strings are copied from here: https://www.raspberrypi.org/blog/new-camera-mode-released/

kostko, 2017-12-08:
Yes, this is what I am saying, do not copy and paste, write it yourself and use the correct `x` symbol (e.g. a standard `x` not for example `×` which is a completely different and unsupported character).

SloMusti, 2017-12-08:
Fixed with 7305bfd

closed 2017-12-08 by SloMusti