IRNAS/PiRA-zero-firmware #13: Webserver bufffer bug

SloMusti, 2017-12-22:
```
22.12.17 11:11:21 (+0100) Exception happened during processing of request from ('192.168.8.168', 50285)
22.12.17 11:11:21 (+0100) Traceback (most recent call last):
22.12.17 11:11:21 (+0100)   File "/usr/local/lib/python2.7/SocketServer.py", line 290, in _handle_request_noblock
22.12.17 11:11:21 (+0100)     self.process_request(request, client_address)
22.12.17 11:11:21 (+0100)   File "/usr/local/lib/python2.7/SocketServer.py", line 318, in process_request
22.12.17 11:11:21 (+0100)     self.finish_request(request, client_address)
22.12.17 11:11:21 (+0100)   File "/usr/local/lib/python2.7/SocketServer.py", line 331, in finish_request
22.12.17 11:11:21 (+0100)     self.RequestHandlerClass(request, client_address, self)
22.12.17 11:11:21 (+0100)   File "/usr/local/lib/python2.7/SocketServer.py", line 654, in __init__
22.12.17 11:11:21 (+0100)     self.finish()
22.12.17 11:11:21 (+0100)   File "/usr/local/lib/python2.7/SocketServer.py", line 713, in finish
22.12.17 11:11:21 (+0100)     self.wfile.close()
22.12.17 11:11:21 (+0100)   File "/usr/local/lib/python2.7/socket.py", line 283, in close
22.12.17 11:11:21 (+0100)     self.flush()
22.12.17 11:11:21 (+0100)   File "/usr/local/lib/python2.7/socket.py", line 307, in flush
22.12.17 11:11:21 (+0100)     self._sock.sendall(view[write_offset:write_offset+buffer_size])
22.12.17 11:11:21 (+0100) error: [Errno 32] Broken pipe
```

kostko, 2018-01-03:
Does this crash the server and/or the application or is just that the traceback is printed and everything continues to work?

SloMusti, 2018-01-03:
Appears to continue working, but there may be exceptions when a reboot of application is required. This bug is less critical at the moment.